AngelLM/Thor #27: Smoothie

arthurwolf, 2016-08-05:
Cool project !

How about using a Smoothieboard and the Smoothie firmware instead of a grbl-based solution ?
We make it very simple to add new "arm solutions" to convert cartesian coordinates into actuator coordinates ( https://github.com/Smoothieware/Smoothieware/tree/edge/src/modules/robot/arm_solutions )
Also, it's 32bits, so it'll be faster and smoother. And it's got a larger community, is very actively developed, etc ...
We'd send you a free board if you say you'll add the code to smoothie.

Cheers :)

AngelLM, 2016-08-05:
Arthurwolf, it would be awesome! 
Of course I'll share everything I'll do with the Smothieboard :)

I have seen that the Smothieboard can control up to 5 steppers, isn't it? As Thor has 6 DOF (without counting the gripper) I think I will need add something to control the last one.

Regards!

arthurwolf, 2016-08-06:
Hey.

Yes, you'll need to add an external driver to it. It's very easy to do, you can use any pin for this.
Can you email me at wolf.arthur@gmail.com with your full shipping address and phone number ?

Thanks :)
Cheers.

AngelLM, 2016-08-18:
Package received!
Thank you very much =)
Let's keep in touch!

closed 2016-08-18 by AngelLM