AngelLM/Thor #29: Malfunction due steppers overheat

AngelLM, 2016-09-26:
Steppers overheat causing printed gears deformation and malfunction of the arm.
**Possible causes:**
1. Wrong stepper drivers or wrong configuration.
2. Not enough refrigeration. Actual design does not allow air flow.
**Possible solutions:**
1. Find other driver that works better.
2. Modify the actual design to allow steppers refrigeration.
3. Use a more heat-resistant material to print the gears: ABS, Nylon, HIPS, PET...

AngelLM, 2016-09-26:
I'll try to solve this keeping the A4988 drivers, modifying the design and trying to print the gears with other materials.

Danny-VdH, 2016-09-27:
I gone use Microstep Driver ST-7128 from stepperonline
My gears are printed with ABSPlus from Stratasys. (If you need some let me know)
When I finished the robot I will give feedback 
I going to look to buildin a ventilation systeem into the second articulation to prevent the heating.

AngelLM, 2016-09-29:
I'll be waiting for that awesome feedback!

I just pushed a new modification of the design, allowing the air-flow and the installation of 40mm fans. Before merging it to the main branch I have to test it.

Danny-VdH, 2016-10-02:
Can't wait to see your solution!

AngelLM, 2016-11-07:
Solved on dd6c5e3!
Yay, the new design enable the air flow and refrigeration of the steppers. Now they don't melt the structure or gears, making possible the arm activity for a long period of time.

Closing this!

closed 2016-11-07 by AngelLM