AngelLM/Thor #31: Tool pins connector slip off.

AngelLM, 2016-09-27:
Pins, actually located on Art4Body piece, are glued to the printed piece. Due to use, glue stopped holding the pins and now it's impossible to connect the tool wires.
Possible solution: Design a small PCB that integrates the tool pins and the endstop as they are in the same side of the piece.

Danny-VdH, 2016-09-27:
I am going to forsee two tubes going through inside the robot, one for the electrical connection on a pcb board with connector and one for use of a suction vacuum system. Both will be connected with flexible tubes to the tool of the robot.

AngelLM, 2016-11-17:
I fixed it with two-part epoxy glue and it works very well... Closing this for now.

closed 2016-11-17 by AngelLM