AngelLM/Thor #32: Feedback sensors

AngelLM, 2016-09-29:
Currently the control of the arm is based on a open kinetic chain. The Zero (Home) position is set at the start and the subsequent movements are based on that reference.
The inclusion of encoders or potentiometers as angle sensors could allow a closed kinetic chain control and new features like recording movements manually.

Danny-VdH, 2016-10-02:
What do you thing about this solution?
http://wiki.seeedstudio.com/wiki/Grove_-_Rotary_Angle_Sensor

Danny-VdH, 2016-10-02:
I'm thinking to build a arduino uno inside the robot base to connect all the potmeters led neopixel ring, buttons and gripper servo motor, then connecting with serial com. to the outside connector to connect to the main (arduino mega board) circuit. It solves a lot of wires (could also be for other electronics to connect with , like fans for controlling the heating inside the robot) And it's a cheap solution!

Danny-VdH, 2016-10-02:
I'm gone make my own connector pcb board that connects over the arduino uno and the pcd boards with the potmeters on it in place of the homing sensors pcb's.

Danny-VdH, 2016-10-02:
You can use a soft serial communication between two arduino's, tested it already and work flawless!

Danny-VdH, 2016-10-03:
http://www.contelec.ch/nc/en/products.html?action=details&pfam=WAL200&typ=Leitplastik&bid=&cHash=b253c401e47bbadd37a18237fb89901b

Hollow shaft potmeter could be a good solution.

Blueicaro, 2016-10-20:
Hi, the industrial robots, from ABB (http://new.abb.com/products/robotics/es), use one resolver for each motor unit. Resolver gives two analogic signal. One signal is called X, and de other is called Y, with this signal you can have polar coordinates of the axis motor, so you have the angle of the axis motor. 
Inside of the robot there is a boad called Serial Measure Board (SMB), this board have connected all resolver unit , read the resolvers and send information to the computer.
Resolver unit, are used to read position, and speed
 This is quite complex and expensive, but I think with the potmeter, as say Danny-VDH you can read the angle of the axis motor.
Using a board (arduino for example) inside the robot, reading the potometer, and sending the information to the main controler can be a good option.
Sending information in serial mode needs some protocol to avoid, and detect transmision errors, but this is easy to build it.
/BlueIcaro
P.D. I'm spanish, and my english is a little rusty, if you want we can discuss it in spanish.

CatalanCaptain, 2016-11-18:
What about hall effect sensor? You can place 3 in a 120º polar array board, and magnets on the other side.
The accuracy is as high as your ADC resolution can be.
I'm spanish too...

ppedro74, 2016-12-12:
@Danny-VdH 

Can you post a supplier to the Wal200 potentiometer ? 
I can't find the price neither were to buy.

Anyone knows the cost ?

i-make-robots, 2016-12-27:
I am using the as5045, same as in MANTIS. ~4K positions double Hall effect. Currently working on making the sensor into a single unit that can be independently tested and then dropped into an arm design where needed.

i-make-robots, 2016-12-27:
I hve seen an arm that uses LEDs, photoresistors, and slots cut in a disc. Measure the light level to find the angle. They claim millions of points per rotation, but do all their math on an fpga.

ppedro74, 2016-12-27:
Can you recommend a board/kit with the as5045 and the magnet ?

i-make-robots, 2016-12-28:
i got one from http://madscientisthut.com/ and tweaked the firmware.  here's what I got: https://www.instagram.com/p/BOim_CBjiaU

ppedro74, 2016-12-28:
Dan,

Thanks for the tip !

i found more development pcb kits (digikey, mouser, farnell)

nice job.

I have the chip glue, tape = mess ...

1) Can you share the 3d print parts and the ball bearings you used ?

2) Are you using that solution with the robot arm ?

i-make-robots, 2016-12-28:
I'm not ready to share teh parts yet, tho I have in the past.  https://hackaday.io/project/945-5-axis-robot-arm  I'm currently trying to update my own arm design based on some good ideas in THOR.

ppedro74, 2016-12-29:
Dan,

Thanks for the feedback.

so far i'm researching potentiometer feedback alternatives. I'm following
you on Twitter, i'll keep an eye to see how your idea evolves.

Maybe i'll try to reproduce something similar my main issue is the board
size, and the space needed to fit in.

One question where you source the ball bearing ?

i-make-robots, 2016-12-29:
FWIW I post more to instagram.  I use a local company that supplies
bearings to industry.  They have a big warehouse an hour away and are nice
enough to tell me which bearings are rare so I don't use some wonky "only 2
available in the world" size.

AngelLM, 2016-12-29:
Wow! I have been a little bit offline in this issue, I really like the several solutions you have been talking about! 
I saw that as5045 has a step size ~ 0.088... awesome! Its kinda expensive but I think it deserves the money. Keep me informed! :)

saeugetier, 2017-01-14:
http://www.distrelec.de/de/winkelsensor-qfn-28-ic-haus-ic-mh-qfn28/p/11039539
This chips seems to be a bit cheaper, but still expensive.

saeugetier, 2017-01-14:
As you can see in the datasheet of the  iC-MH http://www.ichaus.de/upload/pdf/MH_datasheet_C2en.pdf the supported resolution is 12bit (4096 steps per 360 degree). But the accuracy is lower. It depends on the quality of the magnetic target. The nonlinearity (interpolation error) can be reduced by calibration.

AngelLM, 2017-01-16:
Have you seen [uStepper project](http://ustepper.com/shop/home/9-ustepper-premium-kit.html)?
It is licensed under CC-BY-NC and all data of the project is available in its [GitHub repository](https://github.com/uStepper/uStepper)

saeugetier, 2017-01-16:
I know that project. I looked in the schematic. They are using an AS5600 which seems to be quite cheap while having an resolution of 12bits.  I saw this chip on DigiKey for 3€ each @10pieces.
I think that using a feedback for the steppers could result a small improvement. I would be better to have the feedback directly in the joint. In that case a custom feedback electronic has to be designed.

AngelLM, 2017-01-16:
@saeugetier I was thinking about it and makes sense to have the feedback directly in the joint as you said. For hinge joints that could be "easy" to install, but for pivot joints that could be trickier. Also, the last 2 joints would be even harder.
In other hand, having the feedback from motors would ease this job, but due to mechanical reductions we will lost the "absolute" feedback. (Most motors have to turn >360º to do a complete joint revolution).
I'll think about how link a sensor to the pivot joints.

saeugetier, 2017-01-16:
@AngelLM I looked at the mechanics. Adding angle sensors to the robot is quite challenging, but could be feasible. 
I'm skilled in electronics design. If you need help in designing the sensor pcbs, I could assist you.
Another big task would be to do something usefull with the position feedback. GRBL has no position feedback input. GRBL just outputs the steps which each motor has to move. That's why the uStepper has its own microcontroller and motor driver on board. So each uStepper has its own PID closed control loop. 
On Art1 the closed loop is very simple. But the rotation and pitch of Art56 depends on 2 motors. So we cannot just simply clone the uStepper.

Danny-VdH, 2017-01-17:
For the base rotation feedback I'm gonna use a gear with a reduction 1:5 PG5 (the one that's on the geared steppers) parallel agains the driving gear and make a case beneath the reduction with a high resolution potentiometer into it connected with a pigion gear that plugs into the reduction.  The 10 bit analog signal will be having enough resolution 1024/360 degrees =  2.84 at 1 degree. 
Diameter x Pi, (3.14159), divided by 360 for 1 degree. diameter of the Thor robot will approximately be (300mm x 3.14159)/360 =  2.618 / 2.84 = 0.922 mm (I think it will be accurate enough)
Correct me if I'm wrong.

AngelLM, 2017-01-17:
@Danny-VdH long time ago I thought about using potentiometers to get the feedback. I just want to share with you my experience. I used low-cost potentiometers (maybe that was the problem) and I tested them before including in the design. Using an oscilloscope I found that the signal provided by them wa pretty noisy. I calculated how that noise would affect to the feedback and I got around ±2º of precision. Having 6 articulation the error is incremented to... 6 x (±2º) = ±12º at tool. That was why I decided in first place to to an Open-Loop control. I'm sure that your potentiometer won't give that curly signal, but if you have access to an oscilloscope you can test it :)

i-make-robots, 2017-01-17:
I have been playing with AS5045 sensors.  double hall effect, 4096
absolution positions (0.08deg/step).  So far they've been really difficult
to calibrate correctly.  In theory when i get them working right then it
will be a single sensor unit that can be used all over many arms and other
robots.  https://www.instagram.com/p/BO79B5njnhU/

saeugetier, 2017-01-17:
For potentiometer and magnetic encode you can say that resolution and accuracy are different things. The resolution can be 12bits, but due nonlinearities the accuracy could be e.g. >1°. 
For the magnetic encoder an accurate magnet is needed. The size of the north and south pole has to be equal. That's not an easy task. Most of the cheap magnet stores from internet have crappy quality. ASM sells its own magnets. I would recommend using them. The other point is that the magnet has to be placed exact in the middle of the chip. Even though the placing and the magnet are perfect, the resolution will not be the same as accuracy. With help of calibration the accuracy could be improved. But calibration is not an easy task. You will need some kind of reference system.
I think that the accuracy of the robot could be improved by a feedback system. But you should not expect too much...

saeugetier, 2017-02-21:
I saw this cheap encoder on eBay. Maybe it is worth to take a closer look at?

http://www.ebay.com/itm/600P-R-DC-5-24V-6mm-Incremental-Rotary-Encoder-AB-2-phase-Shaft-0-20KHz-NEU/272374834541

Danny-VdH, 2017-02-21:
I did a test with a potentiometer to get the position in a init fase and check every 100 ms to see if a am lossing position while moving the stepper. I must say, it is realy good, cheap and reliable.
For the base movement a use a 10 turns 10K potentiometer. for the arm movements I tested it with a potentionmeter with a hollow shaft. Most of the time it's just for starting up the robot from a power shutdown. I don't have to home my robot positions anymore. :-) Now I'm going to test a mini arduino build into one of the arms and connect all my axis except the base analog on it. Then I'm going to send all data through serial communication to my main controller. I just need 4 wires in my slipring to communicate and power between my main controller. I'm using a slipring between base and the other parts of the robot and this is working very good. Look at my hackaday page to see the photos.

i-make-robots, 2017-02-21:
@Danny-VdH what's your HAD.IO page url?

Danny-VdH, 2017-02-21:
closed loop is another story to do it with potentiometers, but for starting up and knowing your position it is good enough. I don't do anything with my steppers when it stays between 1° tollerance. If I didn't have any obstruction I totally stay working on the stepper counts, it's only when the difference is out of range I rehome my position to the actual potentionmeter position.

Danny-VdH, 2017-02-21:
It's also a good system to check when there was an obstruction to stop the movements of the robot and force it the restart the robot.

Danny-VdH, 2017-02-21:
[https://hackaday.io/projects/hacker/145041](url)

Danny-VdH, 2017-02-21:
I'm going to make a software that reacts when a moving to a particular speed it has to make a distance, if this distance is taking to long I got a following error.

AngelLM, 2017-09-27:
It seems like Danny-VdH made a nice and affordable solution here: https://hackaday.io/project/16665-thor-robot-with-addons-and-gui/log/64709-thor-v202-with-feedback-sensors#
My bad for not noticing it before!
Sounds like solved!

closed 2017-09-27 by AngelLM