AngelLM/Thor #46: Where does the logic lie?

lizixroy, 2017-01-10:
Hi @AngelLM 

First of all, thank you very much for creating Thor, very impressive and helpful indeed.
I have been looking for some control logic (or maybe inverse kinematics solver, etc.) and hardware drivers, but I can't seem to find any actual code in this repository, there are tons of nice assets (CAD files) but the actual programs are no where to be found.

Where do you keep your logic? Maybe this is how a robotic project is typically organized, but as a software developer making his first foray into this field, I am confused by it.

The hardware design side of things is tremendously interesting, but can you write more about the software aspects of the project?

Thank you very much.

BR,
Roy

AngelLM, 2017-01-10:
Hi @lizixroy !

Thanks for your kind words :) 
For the control of the motors I made a modified version of GRBL. You can find the Thor's Firmware in [this repository](https://github.com/AngelLM/grbl). I have not developed GRBL, that's why I thought that this modification should be a fork from the original one.

Right now I use GRBL, flashed inside an Arduino Mega, to control 7 steppers using forward kinematics. I have solved the inverse kinematics, but when I tried to put it into the GRBL inside the Arduino Mega didn't work. Probably the processor was not powerful enough. 
There are users at Hackaday community who are developing awesome software for Thor, I recommend you to take a look to [Danny's posts ](https://hackaday.io/project/16665-thor-robot-with-addons-and-gui#menu-logs)

Hope it helps, feel free to ask anything you want to know :)

williamleferrand, 2017-01-14:
Hi there, 

Thanks @AngelLM for your work - I'm building a Thor too; my goal is to make it remote-controllable via the internet and to pair it with a printer so that remote users can print+assemble simple designs. 

I looked at Danny's UIs and I'm probably going to do a web based version of his manual UI in the coming days - I'll share it on github!

AngelLM, 2017-01-14:
Hi @williamleferrand !
That would be an amazing project! Keep me informed :)

AngelLM, 2017-01-14:
As the main question was solved, I'm closing the issue.

closed 2017-01-14 by AngelLM

williamleferrand, 2017-01-14:
@AngelLM will do! did you get my email btw?

AngelLM, 2017-01-15:
@williamleferrand Let me check it right now!