AngelLM/Thor #48: Readme: Printer Requirements

saeugetier, 2017-01-14:
Could you mention in the readme which kind of printer was used to print Thor, please?
Is 200x200x180 enough space to print all parts?

williamleferrand, 2017-01-14:
It is if you use the stls that are in stl/modifications (for printing bits of Art2 & the base). Otherwise according to the hackaday post, 

"The most important fact is that the largest printable piece designed is 290mm of depth and I recommend to print it lied flat, that's why I suggest using a printer that has at least 290mm of printing length on one of its horizontal axis."

AngelLM, 2017-01-15:
Thanks @williamleferrand !
@saeugetier , as William said you can print the modificated parts with your printer. I'll add this question to the FAQ.

closed 2017-01-15 by AngelLM