AngelLM/Thor #53: Art56 Small Gears do not fit all variations of GT2x40 pulleys

b-adkins, 2017-03-31:
It's a clever use of a set screw to attach the pulley to the 3d printed gear! However, it looks like the GT2 standard doesn't apply to the set screws. The pulleys I ordered ended up having 4mm (not 3mm) set screw holes that do not line up with the hole in the gear.

So I used hot glue. I'll see if it can handle the torque.

AngelLM, 2017-04-03:
Hi @b-adkins ,
I didn't knew it, I assumed that it was standard. Anyway, you can adjust the 3D model to fit to your needs, if you need help with the modification of that 3D model tell me ;)
Also, I'm interested in knowing if the hot glue fix work with a load of 750g!

AngelLM, 2017-05-09:
@b-adkins how it's going? the hot glue worked?

closed 2017-08-08 by AngelLM