AngelLM/Thor #64: cinematic motion algorithm

mkeyno, 2017-08-04:
hi, sorry to open ticket for my question , but actually I'm little lost to find the methods can convert input location (Cartesian , polar) to the optimized cinematic motions of each axis? for example how to defined the  best path from  A to B coordinates and how it converted  the cinematic  motion for each axis? what is the optimization function & type ?how to define cinematic motion parameters of specific arms? I really appreciated for your hint

i-make-robots, 2017-08-04:
what do you mean when you say "cinematic" motion?

i-make-robots, 2017-08-04:
Thor does not, at this time, afaik, have a motion planning software that would, say, let you plot lines in space and then move the arm to meet those points, like this: https://www.youtube.com/watch?v=AO57-fZUWho

mkeyno, 2017-08-05:
sorry I mean  kinematic , you know, every robot has its own physical parts and degree of freedom so to define path between two given points we need the calculate some  dynamic equations to find this patch and other parameters(like velocity and acceleration ) , however there is no unique path between two given points , thus some optimization algorithm applied to find the best and smooth path , I saw couple of repo use the CNC planner code to implemented this path but not find yet the path optimization algorithm

i-make-robots, 2017-08-05:
Frequently the shortest path is taken.  I know the number of steps each motor must take and the speed of the slowest motor.  I then step all motors so that they arrive simultaneously at the destination.  There are times when this solution is bad - it does not take into account impossible geometries or singularities BUT if you can avoid them it works great.

i-make-robots, 2017-08-05:
sorry, to calculate the kinematics is how you get the number of steps of each motor.  I have some examples in https://github.com/MarginallyClever/Robot-Overlord-App/ for 4 and 5 DOF arms.  Maybe that will help?  I don't remember atm if THOR is supported yet, but I'd like to see it happen.

AngelLM, 2017-08-07:
Hi! Sorry for the late reply.
Currently Thor moves using Forward Kinematics (at least mine). The calculations involved in the movement are responsibility of the firmware (GRBL). At this time there is not implemented neither a control using Inverse Kinematics nor movement optimization.
If it helps here it's the [solution for the Forward Kinematics](https://hackaday.io/project/12989-thor/log/43941-forward-kinematics). And here the [solution of the Inverse Kinematics](https://hackaday.io/project/12989-thor/log/44018-inverse-kinematics) of Thor.
Also, @Danny-VdH were developing a GUI to control robotic arms. You can chek it out too ;)

Hope it helps!

i-make-robots, 2017-08-07:
If I have the relative distance between each joint in the system and an simplified STL of each THOR part I could add it to Robot Overlord.  Then it would send FK commands to the arm, much as it does with my other robots.  If that's of interest to you, make it so.  I could get it implemented before the end of August.

AngelLM, 2017-08-07:
@i-make-robots That sounds good! I think that you have all the distances you need in [this image](https://github.com/AngelLM/Thor/blob/developer/doc/ThorDimensions.png).

i-make-robots, 2017-08-07:
I explained myself badly.  I need an STL of each section - the base, the
shoulder, the bicep, the ulna, the wrist.  For each part, I need the origin
of the part to be at the point of rotation or, in the case of the base, at
the world origin.  For each part I also need a measurement of the distance
from the origin to the axis of rotation of the next part.  In the case of
the base, I need to know how far up to reach the point where the shoulder
rotates.  the shoulder origin would be at that spot, and the relative
distance from there to the rotation point of the bicep, and so on.  Does
that make more sense?

I can spend hours monkeying around trying to fix it in software but I've
got very limited time. I'm hoping we can help each other, here.

AngelLM, 2017-08-08:
Hi @i-make-robots 
I just upload the [STL files of each articulation](https://github.com/AngelLM/Thor/tree/developer/stl/FullArticulations)
Also, the distances are the following, understanding global positions as the XYZ coordinates of the space and the local position as the XYZ coordinates of the piece.

- **Base**
    - Global position (X,Y,Z): 0,0,0
    - Rotation axis: None
    - Rotation axis local position: None
- **Articulation 1**
    - Global position (X,Y,Z): 0,0,49
    - Rotation axis: Z
    - Rotation axis local position: 0,0,0
- **Articulation 2**
    - Global position (X,Y,Z): 0,0,137
    - Rotation axis: Y
    - Rotation axis local position: 0,0,65
- **Articulation 3**
    - Global position (X,Y,Z): 0,0,295
    - Rotation axis: Y
    - Rotation axis local position: 0,0,64
- **Articulation 4**
    - Global position (X,Y,Z): 0,0,377,5
    - Rotation axis: Z
    - Rotation axis local position: 0,0,0
- **Articulation 5&6**
    - Global position (X,Y,Z): 0,0,510
    - Art 5 Rotation axis: Y 
    - Art 5 Rotation axis local position: 0,0,47
    - Art 6 Rotation axis: Z
    - Art 6 Rotation axis local position: 0,0,0

I don't really know if the STL of the last 2 articulation will work (because of coupled movements). Probably splitting it into 2 pieces (the column + gears for the 5th and the rest for the 6th) would make the work. I'll do it now, just in case.

AngelLM, 2017-08-08:
Split done.
The new distances would be:

- Articulation 5
    - Global position (X,Y,Z): 0,0,510
    - Art Rotation axis: Y
    - Art Rotation axis local position (X,Y,Z): 0,0,47
- Articulation 6
    - Global position (X,Y,Z): 0,0,610.431
    - Art Rotation axis: Z
    - Art Rotation axis local position (X,Y,Z): 0,0,0

i-make-robots, 2017-08-08:
are you sure about articulation 2, distance 65?  looking at the parts
together it looks more like 85 or 90.

i-make-robots, 2017-08-08:
> Ph: +1 (604) 259-9564 <(604)%20259-9564>
> Mo: +1 (604) 916-2281 <(604)%20916-2281>

ok, I have the model displaying after adjusting the origins.  You'll
probably want to split the gripper tool from the wrist component.  Biggest
reason so you can animate the gripper action.  I'll take some more time
later this week to implement FK.

i-make-robots, 2017-08-08:
aaaand now i have FK working, kind of.  Fingers crossed I can get IK
working just as easily.

i-make-robots, 2017-08-08:
Right, IK looks to be mostly working.  I've done the first 95%, please
contribute whatever you can to make it even better.

https://www.instagram.com/p/BXi0Ci4gLqb/

AngelLM, 2017-08-08:
You are right! Is Z 95mm

AngelLM, 2017-08-08:
I saw it! Awesome work! 
Expect some contributions and feedback from my side! :)

closed 2021-08-09 by AngelLM