AngelLM/Thor #67: ART2 BodyA that meshes with two gears always slipping. Bad fit?

martinbarr, 2017-11-17:
Hi Angel,

After some serious testing and trials in getting the robot arm to move, it worked for a little while then I started having serious slipping issues. I watched and saw the ART2 A arm flex too much at the bottom where the only support it has are the two pins on the top motor and the ART2 Union. Not sure how everyone else has got on with this or if they have also had issue too. Would love to know. I will look to try and solve my particular issue,

Thanks in advance

Martin Barr

martinbarr, 2017-11-22:
Hi All,
I logged this issue and carried on investigating.

After several attempts to correct I found this issue that I thought I would share. Maybe I missed the depth that the pulley wheel should be. I did a sketch to show what I did.

[image]

AngelLM, 2017-11-24:
Hi @martinbarr !
Do you think that it's possible that the pin in the opposite side is not longer enough? This could be causing this articulation to be a bit loose.

Other reason could be a bending issue... Maybe the Articulation 2 is not acting as rigid piece and the lower parts are bending outwards... I don't really know.

AngelLM, 2018-04-25:
Did you solved it?

closed 2019-01-31 by AngelLM