AngelLM/Thor #68: Some dimensions that may be wrong

diegotrap, 2018-01-23:
Hello everyone,

I am printing a complete set of Thor's printed parts and while testing the fit of the different parts with bearings, bolts, etc I have noticed some dimensions that make assembly very hard or require the printed part to be postprocessed (sanded, drilled, etc.).

This notes refer to the DesignUpdate branch, which is the one that I am using.

These are the ones that I have caught by now:

* **Gripper**: every bolt head pocket have a radius of 2.7mm. M3 DIN912 bolts have a 2.75 nominal radius, so the heads won't fit. Other bolt head pockets in Thor have 2.95mm radius. Changing the Master bolt head won't be enough since not every head references to the entity, it has to be done one hole at a time.
* **BaseBoxBody**
    * fan bolt head pocket have a radius of 2.7mm. Changing the Master bolt head won't be enough since not every head references to the entity, it has to be done one hole at a time.
    * insert holes are too big for 5mm inserts. Holes should be at least a couple of tenths smaller to hold the insert.
* **Art56GearPlate** inner bore for a 625 bearing is only 16mm in diameter, some slack has to be included to be able to push the bearing in.
* I find **BearingFix** parts hard to get into their pockets, I would reduce the diameter by some tenths.
* **BearingPlug** won't fit in it's hole. Maybe is meant to be printed in flexible material?

Other stuff:
* This is a FCSTD1 file that has made his way into the repo: https://github.com/AngelLM/Thor/blob/DesignUpdate/freecad-src/Art2SideCover.FCStd1

Sorry not to be able to change it myself, my set of files is already very different from the source files and my git knowledge is not very extensive.

I'll post here if I spot more things like these.

Diego.

AngelLM, 2018-01-25:
Thanks Diego for the feedback! I'll check it ;)

AngelLM, 2018-01-25:
Okay, this are the changes I did until now
- **Gripper**: Bolt heads modified from 2.7mm to 2.95mm
- **BaseBoxBody**: Bolt heads modified from 2.7mm to 2.95mm, holes for the inserts modified from 5mm to 4.8mm.
- **Art56GearPlate**: Inner bore modified from 16mm to 16.2mm

About the *BearingPlug* piece: it was not designed to enter completely into the hole, only a bit,to prevent the bearing balls from pop out the rail. Maybe the plug piece diameter is larger than it should be. I'll check it.

Finally, *BearingFix* is a complicated piece to modify as it is already small. The problem is only the diameter or it is alse the fixing tabs?

AngelLM, 2018-07-04:
Solved and merged!

closed 2018-07-04 by AngelLM