AngelLM/Thor #69: Articulations bodies

LoLo2207, 2018-02-06:
Hi, i'm printing all the parts and I only have left the Art1Body, Art3Body but each will take +24h to print because they need supports. I live with more people and can't print at nights.

Could you please tell me what settings did you use?
Or could you modify them to print them in several parts?
Thank you.

Danny-VdH, 2018-02-06:
You can download them in pieces from my hackaday page.
https://hackaday.io/project/16665-thor-robot-with-addons-and-gui
Best regards.

AngelLM, 2018-02-06:
Hi @LoLo2207 

@Danny-VdH did an awesome work splitting the big pieces of Thor! As he did, I suggest you to follow the previous link and download that pieces  ;)

closed 2018-02-06 by AngelLM