AngelLM/Thor #72: Looking for a Print Inventory List, and unit scaling on STLfiles.

AlexanderPeters, 2018-10-06:
Hi, could you please let me know where I can find a Bill Of Materials that includes the quantity of each number of 3D printed component that need to be printed. Additionally can you please let me know if the .stl files are scales in in, mm, or cm.

-Thanks

AngelLM, 2019-02-07:
Hi @AlexanderPeters ,

Currently there is not a BOM including the quantity of printed pieces. I'll do it soon. Meanwhile you can download the FreeCAD assembly file and check it.
The unit used on the STL files, and every 3D file in this repository is mm.

Thank you for your suggestion!

AngelLM, 2021-08-09:
In the new [Thor's website](http://thor.angel-lm.com/) I've added a 3D Printing Guide with the ammount of pieces needed of each model and the recommended settings to print them all.

closed 2021-08-09 by AngelLM