AngelLM/Thor #76: Alimentación del servo de la articulación 6

lisergio, 2019-06-20:
En primer lugar me disculpo por escribir en castellano ( Traducción después con google)...

Estoy montando la electrónica "ControlPCB" siguiendo el esquema, y me he fijado en que el servo va conectado directamente sin ningún regulador de tensión... por lo que imagino que lo estamos alimentando con los 5v de la placa Arduino Mega, es correcto??

Gracias.
Gran trabajo, 

estoy esperando a que me lleguen todas las piezas mientras imprimo todo...

Traducción :

First of all I apologize for writing in Spanish (Translation later with google) ...

I am mounting the electronics "ControlPCB" following the scheme, and I noticed that the servo is connected directly without any voltage regulator ... so I guess we are feeding it with the 5v of the Arduino Mega board, is it correct? ?

Thank you.
Great job,

I'm waiting for all the pieces to arrive while I print everything ...

VictorLamoine, 2019-06-21:
Do NOT connect anything that sources power to an Arduino pin, you will most probably fry the Arduino.
Read this: https://arduino.stackexchange.com/questions/88/is-there-a-limit-on-how-much-current-a-pin-can-sink.

lisergio, 2019-06-21:
Gracias, eso lo entiendo.

pero en el esquema del circuito no hay ningún regulador de 5v, por lo que el servo se alimenta de la placa Arduino.

Thanks, I understand that.

but in the circuit diagram there is no 5v regulator, so the servo is powered by the Arduino board.

[image]
[image]

hay que añadir una alimentación de 5v solo para el servo?
should we add a 5v power supply only for the servo?

gracias
thanks

AngelLM, 2019-06-24:
Hola @lisergio 
Como bien dices, el cable de alimentación del servo va conectado a los pines que has marcado en las imágenes. Las salidas de VCC y de 5V auxiliares están conectadas con la salida de 5V de Arduino (y como este ya incorpora su propio regulador de tensión no hay que añadir nada más).
En resumen, puedes conectar el servo directamente a la placa sin problemas.
¡Espero haberte ayudado!

Hi @lisergio, as you said, the servo's power wire is connected to the pins that you marked in the images. The VCC and 5V aux outputs are connected to the Arduino's 5V Output (and as it already has its own power regulator, there is no need to add anything to the circuit).
To sum up, you can connect directly the servo to the board without any problem.
Hope it helps!

lisergio, 2019-06-25:
Eso era justo lo que quería saber, muchas gracias!!
La cuestión es que voy a usar un servo con piñones metálicos (20kg , 0,15seg) y el consumo es un poco alto...
Así que tendré que colocar una fuente aparte de 5v solo para el servo y así  proteger el regulador de la placa arduino.

That was just what I wanted to know, thank you very much !!
The point is that I'm going to use a servo with metallic pinions (20kg, 0.15sec) and the consumption is a bit high ...
So I will have to place a separate source of 5v only for the servo and thus protect the controller from the Arduino board.

closed 2019-06-25 by lisergio