AngelLM/Thor #77: Duda en el montaje / Doubt in the assembly

lisergio, 2019-07-05:
hola de nuevo...
he empezado el montaje de thor, y tengo una duda...
las siguientes piezas forman parte del mismo montaje?
Art3TensionerBody.stl
Art3TensionerPulley.stl
2	×	MF84ZZ Bearing
2	×	14mm x ∅5mm Rod

el paso del rodamiento es de 4mm, y la varilla es de 5mm, por lo que no encaja.

Hello again...
I have started the assembly of thor, and I have a doubt ...
the following parts are part of the same assembly?
Art3TensionerBody.stl
Art3TensionerPulley.stl
2 × MF84ZZ Bearing
2 × 14mm x ∅5mm Rod

the bearing pitch is 4mm, and the rod is 5mm, so it does not fit.

AngelLM, 2019-07-05:
Hola @lisergio 
Todas esas piezas forman parte del mismo montaje, pero el diámetro de las varillas de 14mm de largo debería ser de 4mm en lugar de 5mm. En la [lista de materiales ](https://github.com/AngelLM/Thor/wiki/Component-list) aparece así, ¿lo he publicado mal en otro lugar?
¡Un saludo!

Hi @lisergio 
All of that pieces are part of the same assembly, but the diameter of the 14mm length rods should be 4mm instead of 5mm. As it appears in the [list of materials](https://github.com/AngelLM/Thor/wiki/Component-list) Have I published it wrong somewhere?
Regards!

lisergio, 2019-07-05:
en la lista de materiales de hackaday.io pone que son de 5mm

in the hackaday.io list of materials says that they are 5mm

Gracias!!! Aclarado!!

AngelLM, 2019-07-05:
Corregido en el post de Hackaday, gracias por darte cuenta!

Fixed in the Hackaday post, thanks for find it out!

closed 2019-07-05 by AngelLM