AngelLM/Thor #78: Servo not move... / El servo no se mueve

lisergio, 2019-07-24:
hola de nuevo, ya tengo varias articulaciones montadas, y he empezado a probar la electronica, pero tengo un par de problemas... todos los motores se mueven, aunque no todos en los grados que deberian ( he usado tu configuracion de GRBL ) , será cuestion de modificar los pasos/mm , pero la pega es, que usando Asgard, la pinza, no rota, el boton de la articulacion 5 y de la 6 hacen lo mismo, y por ultimo el servo no hace nada... 
he probado con otro por si a caso y tampoco funciona, pero si hace un pequeño movimiento al iniciar el sistema. 

Gracias! 

hello again, I already have several assembled joints, and I have started to test the electronics, but I have a couple of problems ... all the motors move, although not all in the degrees that they should (I have used your GRBL configuration), it will be a matter of modifying the steps / mm, but the glue is, that using Asgard, the clamp, does not rotate, the button of the articulation 5 and the 6 do the same, and finally the servo does nothing ...
I've tried another one just in case and it doesn't work, but it does make a small movement when starting the 

Thanks

lisergio, 2019-07-26:
He solucionado el problema del servo, ya se mueve, era un problema del conector... pero no me hace todo el recorrido, donde se puede ajustar?

en cuanto al giro del gripper, sigue sin funcionar usando Asgard  ( enviando un Gcode,Y90 Z-90 si funciona)

I have solved the servo problem, it is already moving, it was a connector problem ... but it does not make me the whole route, where can it be adjusted?

As for the rotation of the gripper, it still does not work using Asgard (sending a Gcode, Y90 Z-90 if it works)

mnc5, 2019-07-27:
unsubscribe   please

closed 2019-07-27 by lisergio