AngelLM/Thor #90: Payloads

x0rium, 2021-11-19:
Hi! Does the project support laser engraving or 3d printer heads connection? I found only cnc mill head connection sample.

AngelLM, 2021-11-20:
Hello! As far as I know no one has mounted a laser engraving or a 3D printer nozzle to Thor yet, but it sounds very interesting!
In terms of mechanics, you could design a gripper for the tool you want to mount based on the Art56Interface piece.
As for the electronics, the current control board as a digital PWM output (f.e. to control a laser) and a spare stepper driver that could control the motor of a extruder... So I don't think it would be impossible to achive, but it would be a very dedicated job :)
Just curious, where have you seen that cnc mill head connection example?

Best regards!

x0rium, 2021-11-23:
Thx for answer :) 
https://youtu.be/_EjADbIS9GQ
I think it's will be grate to use roboarm as universal mechanics with functional heads. For example to making pcb
1) textolite panel with photoresist film illuminated by an ultraviolet laser
2) with the help of a manipulator it is transferred into a bath with a chemical composition to remove the film
3) drilling holes   
4) pick and place elements 
5) move to hot plate 
6) testing
Sounds very cool :)

x0rium, 2021-11-26:
https://github.com/madl3x/x-scara

closed 2021-11-27 by x0rium