AngelLM/Thor #91: Confusion on PCB circuit

DigeshC, 2021-11-23:
Hi! We are planning to build Thor but had some confusion regarding the PCB circuit. We were confused on what P1...P8 were on the circuit diagram and how they are connected in the circuits. Thank you so much!

AngelLM, 2021-11-24:
Hello!

The Control PCB is pretty similar to the RAMPs 1.4 and this feature is a legacy of this board.
P1...P8 are groups of 3 columns of 2 male pins. One pin of each column is connected to 5V and the oposite pin is connected to the driver's microstepping pin (MS0, MS1 and MS2 respectively (See [PCB Control Schematic](https://github.com/AngelLM/ThorControlPCB/blob/master/Docs/ControlPCB-Schematic.pdf)).
MS pins are used to configure the microstepping by setting the pin to high level (5V), shortcutting the columns using jumpers (See [B section of this image](https://www.researchgate.net/publication/339101918/figure/fig4/AS:867963263528965@1583949976723/Wiring-of-XYZ-System-A-RAMPS-14-shield-top-and-Arduino-Mega-bottom-B-RAMPS-14.png)). 
Depending on which columns have been short-circuited the configuration will be different (See [microstepping configuration table](https://i.stack.imgur.com/vN7JL.png)]
After setting the jumpers, you can place the stepper driver on top of them.

Hope it helps!
Ángel

DigeshC, 2021-12-17:
Hello!
Thank you so much for your explanation. It cleared up our confusion.

closed 2021-12-17 by DigeshC