AngelLM/Thor #105: Questions about rods specification

nomorewzx, 2024-08-31:
Hi, I am now trying to figure out the [BOM list](https://github.com/AngelLM/Thor/wiki/Component-list) 

I see the 2 rods with specification: 

1. Length 102mm, Diameter 5mm
2. Length 128mm, Diameter 5mm

However in the Aliexpress link of the rods, I found that the maximum length of the rods with 5mm is actually 100mm.

I tried search other rods in Aliexpress and could not find the 5mm-rods with length 102mm or 128 mm.

My questions are: Does 100mm x 5mm rods works as well? If not, is there updated rods link? 

Thanks a lot

AngelLM, 2024-08-31:
Hello @nomorewzx !

First of all, I would recommend you to have a look at the [project website](http://thor.angel-lm.com), where you will find much more updated information, as the repository wiki is outdated.
That said, in the[ BOM section of the website](http://thor.angel-lm.com/documentation/bom/) you can find links to the sticks. Because the measurements required for the construction of Thor are very specific, you have to buy longer rods and cut them to the required length.
If you don't have the necessary tools, as the rods are of a very common diameter you may be able to find them at a local hardware store where they can help you with the cutting.

Finally, if you are going to embark on the construction of your own Thor, I would recommend you to join the [project forum](http://thor.angel-lm.com/forums/), where you can share your construction process, ask questions and see other users' doubts. We also have a [Discord server](http://thor.angel-lm.com/forums/topic/thor-discord-server/) for more direct communication.

Hope this helps!

Translated with DeepL.com (free version)

AngelLM, 2024-08-31:
Hello @nomorewzx !

First of all, I would recommend you to have a look at the [project website](http://thor.angel-lm.com/), where you will find much more updated information, as the repository wiki is outdated.
That said, in the[ BOM section of the website](http://thor.angel-lm.com/documentation/bom/) you can find links to the sticks. Because the measurements required for the construction of Thor are very specific, you have to buy longer rods and cut them to the required length.
If you don't have the necessary tools, as the rods are of a very common diameter you may be able to find them at a local hardware store where they can help you with the cutting.

Finally, if you are going to embark on the construction of your own Thor, I would recommend you to join the [project forum](http://thor.angel-lm.com/forums/), where you can share your construction process, ask questions and see other users' doubts. We also have a [Discord server](http://thor.angel-lm.com/forums/topic/thor-discord-server/) for more direct communication.

Hope this helps!

closed 2024-08-31 by AngelLM